Mayitzin/ahrs #2: error "Syntax Error" During install

massuncao0516, 2019-12-12:
I have the follow error after the install command; sudo -H pip install ahrs
Collecting ahrs
  Using cached https://files.pythonhosted.org/packages/37/fd/41a29f8f491b0b87634a874715f56f6bdaa49202e1701e1c588bbf6129dd/AHRS-0.1.2.post3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-Tf5165/ahrs/setup.py", line 22, in <module>
        from ahrs import get_version
      File "ahrs/__init__.py", line 11, in <module>
        from . import common
      File "ahrs/common/__init__.py", line 8, in <module>
        from . import orientation
      File "ahrs/common/orientation.py", line 619
        R = rotation(axes[i], angles[i])@R
                                                       ^
    SyntaxError: invalid syntax
    
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-Tf5165/ahrs/

There is an errir in @ symbol in the code.
I'm using the python3.6 

Thanks

Mário Assunção

Mayitzin, 2019-12-16:
I tried to replicate the error with Python 3.6.8, but couldn't get it.

The mentioned line with SyntaxError is different in the source code, even in the cached version that the link points to. The correct line should be:

```python
R = rotation(axes[i], angles[i])@R
```

that is with an upper-case **R** at the end. I assume this would immediately fix it. I don't really know where you are getting a modified version of it.

I suggest you clone the current version of the repository and install from it:

```
git clone https://github.com/Mayitzin/ahrs
cd ahrs
pip install .
```

If you keep facing problems let me know and I'll try to attack them.

massuncao0516, 2019-12-18:
Hi Mayitzin,

The lowercase R is a copy-paste bug, because it was a uppercase R. 
My guess is: the pip install is not use the python version 3.6, instead use the 2.7, native version from raspberry system. When I run the command "python setup.py install" inside the arhs folder, the error is not the same because the python command is version 3.6.
Now I have several errors due not installed libraries used in your code, Is a problem in my raspberry system. 
Thanks for you attention. 
Mário

Mayitzin, 2019-12-18:
If that is the case try to install it using `pip3`, maybe that would avoid some of those problems.

On the other hand, `ahrs` uses only numpy and scipy, but I assume the package giving some troubles would be scipy. It is used merely to load `mat` files, but perhaps I can find a workaround to eliminate its need in the future.

closed 2019-12-18 by Mayitzin