Mayitzin/ahrs #5: Does this package do position estimation?

stormyweathers, 2020-04-28:
The readme specifies that this is used for position and orientation estimation, but the filters seem to only estimate orientations.

Can I use this package for 6-DOF position/orientation estimation?

I am new to AHRS/MARG stuff so please be patient,

Mayitzin, 2020-04-29:
Hi! It's always good asking and this is not a minor question.

Yes, the package can do orientation estimation. The submodule `filters` provides several algorithms to do so. I'd recommend the `Mahony` and `Madgwick` ones. They are the most common used everywhere.

For example, having gyorscope, accelerometer, and magnetometer in different NumPy arrays, you can obtain the quaternions (orientation) with:

```python
num_samples = gyorscope.shape[0]
q = np.tile([1., 0., 0., 0.], (num_samples, 1))
madgwick = Madgwick()
for i in range(1, num_samples):
    q[i] = madgwick.updateMARG(q[i-1], gyorscope[i], accelerometer[i], magnetometer[i])
```

If you don't have any magnetometer information, you simply feed accelerometer and gyroscope information and should still work.

```python
for i in range(1, num_samples):
    q[i] = madgwick.updateMARG(q[i-1], gyorscope[i], accelerometer[i])
```

I'm working on easier and more intuitive ways to improve this. But the current version works as above.

stormyweathers, 2020-05-01:
Thanks for the reply!

To be specific though, this package does not do position as well?

Mayitzin, 2020-05-03:
Not for this case. The estimation of position would require further references about the surroundings. You can estimate it purely using intertial measurements, but that requires a larger model based on your needs.

stormyweathers, 2020-05-04:
Ok that makes sense.  I will try to work off your example and extend to a full pose estimation for my application.

Before I get started, I have one more question.  In the madgwick filter example, it appears that the IMU estimate is much better than the MARG estimate.  This is surprising since the MARG estimate has more sensor data.

Is this a bug or am I missing something?

This output is generated using the example data from the repo.
[image]

Mayitzin, 2020-05-12:
Since both IMU and MARG have distinctly different architectures, the filter will weight them differently. That means, for IMU you need to _tune_ the gain to a certain value, and a different one for MARG. In this example, I tuned the gain for an IMU, which yields the best results for it, but not for a MARG.

Mayitzin, 2021-02-01:
Closing the issue, as it seems to have cleared the point in question.

closed 2021-02-01 by Mayitzin