Mayitzin/ahrs #7: NED to ENU frame - quaternion math

geoKinga, 2020-08-18:
Hi,
I'm using vectornav IMU fo measurements, here is its math:
https://www.vectornav.com/resources/attitude-transformations

This inertial unit allows to record measurement (accelerometer, gyro, magnetometer), also the unit is returning the orientation angles (yaw, pitch roll) and the quaternions.

So I did try to use measurement values from this unit and calculate (by the use of Madgwick filter from your repo) orientation angles and quaternions. But when I try to compare the quaternions on output from Madgwick filter (your repo) to quaternions on output from vectornav IMU,  the results completely differ.

I know that vector nav is using NED reference frame, can you please tell me something about the frame which is used in your algorithm. Like describe here: https://www.vectornav.com/resources/attitude-transformations

I did try to do a transformation NED > ENU, following to this:
https://stackoverflow.com/questions/49790453/enu-ned-frame-conversion-using-quaternions
but it seems to be the wrong solution.

Can you please give me tips on what I'm doing wrong?
Will'be very grateful for any support with this problem.

PS. Here is my data (in mat file) and  python code which I was used:
https://wutwaw-my.sharepoint.com/:f:/g/personal/kinga_wezka_pw_edu_pl/EgtvnRx3aF9OvkOpsFCnlOwB_pO8ebbyv9OAmezrGNdjPQ?e=pk0oPQ

Best Regards,
Kinga

Mayitzin, 2020-08-20:
Hi Kinga,

I think it lies on the format of the angular rate given by the gyroscopes. It must be given in radians/second. So, you might want to transform it first to radians.

Try that and let me know if it improves for you.

Longhuiberkeley, 2020-12-10:
Hi, @Mayitzin. I have a related question, what's the default input for the IMU sensor in this lib? Like the right-hand Z-top system (XYZ)? Thanks

Mayitzin, 2020-12-10:
Hi, yes, the main reference frame of the package follows the right-hand convention as you mentioned.

Although for aircrafts the left-handed system is preferred, personal navigation systems use the right-hand system and, therefore, this is also used here.

Mayitzin, 2021-05-18:
Closing this issue, due to inactivity, and apparently was solved with the recommended measures.

closed 2021-05-18 by Mayitzin