Mayitzin/ahrs #8: TypeError: 'module' object is not callable

mathang, 2020-09-24:
Looks like a very interesting toolkit, thanks for sharing it. I've tried to run the Mahony AHRS algorithm on some accel and gyro data I have but keep running into this issue:

Traceback (most recent call last):
  File "C:/Users/Ross/AHRS Python Mahony Test.py", line 16, in <module>
    orientation = Mahony(gyr=gyro_data, acc=acc_data, frequency=50.0)
TypeError: 'module' object is not callable

My code is below, I'm sure I'm missing something I'm just not sure what it is. If you have any info that would be appreciated. 

`
import numpy as np
import pandas as pd
from ahrs.filters import mahony as Mahony

num_samples = 20000 #I have chosen this value just to test. Should be array length

acc_data = pd.read_csv('accel3.csv')
gyro_data = pd.read_csv('gyro3.csv')

orientation = Mahony(gyr=gyro_data, acc=acc_data, frequency=50.0)

orientation = Mahony()
Q = np.tile([1., 0., 0., 0.], (num_samples, 1)) # Allocate for quaternions
for t in range(1, num_samples):
    Q[t] = orientation.updateIMU(Q[t-1], gyr=gyro_data[t], acc=acc_data[t])
    
`

Mayitzin, 2020-09-24:
Hi, change the call to the estimator to:

```
from ahrs.filters import Mahony
```

This way you'll import the class Mahony handling the estimation, and not the submodule with the routines.

mathang, 2020-09-24:
Thanks for the quick feedback, I've changed it as you've written. I had changed it to "mahony as Mahony" because I couldn't get it to run previously. The error shows this:

/////

Traceback (most recent call last):
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\pandas\core\indexes\base.py", line 2897, in get_loc
    return self._engine.get_loc(key)
  File "pandas/_libs/index.pyx", line 107, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 131, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1607, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1614, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/Ross/AHRS Python Mahony Test.py", line 21, in <module>
    Q[t] = orientation.updateIMU(Q[t-1], gyr=gyro_data[t], acc=acc_data[t])
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\pandas\core\frame.py", line 2995, in __getitem__
    indexer = self.columns.get_loc(key)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\pandas\core\indexes\base.py", line 2899, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 107, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 131, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1607, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1614, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 1

/////
The data format is 2 x csv files with three columns for the X,Y,Z axis of the accel and gyro. It is loading fine into Python. I am in Australia so it does use . for decimals and , for spacing if that matters?

       0.000  0.000.1  0.000.2
0       0.000    0.000    0.000
1       0.000    0.000    0.000
2       0.000    0.000    0.000
3       0.000    0.000    0.000
4       0.000    0.000    0.000
...       ...      ...      ...
355169 -0.069   -0.006   -0.147
355170 -0.072   -0.001   -0.150
355171 -0.059   -0.000   -0.134
355172 -0.051   -0.011   -0.118
355173 -0.043   -0.019   -0.111

[355174 rows x 3 columns]

Any help to get this running would be greatly appreciated!

Mayitzin, 2020-09-25:
It could be that the data isn't properly formatted, but apparently pandas ld recognize it correctly and loaded it fine.

Can you check the size of the array? You can easily do it with the property `shape` of your pandas data. In order to work with the estimators, each sensor data should be of size N-by-3, where N is the number of rows (observations.)

A second problem might be the contents of the data. If your accelerometers (or magnetometers) have observations equal to zero, they would create a problem dividing by zero, which is not allowed.

I hope this can help you.

mathang, 2020-09-27:
Thanks for your help - still not working though sorry... I've removed all 0 values from the file, and checked the shape and it all seems fine:

0       0.000001   0.000001    0.000001
1       0.000001   0.000001   -0.000001
2      -0.000001   0.000001    0.000001
3      -0.000001   0.000001   -0.000001
4      -0.000001  -0.000001    0.000001
...          ...        ...         ...
355169 -0.068939  -0.005798   -0.147339
355170 -0.071991  -0.001007   -0.150238
355171 -0.059143  -0.000397   -0.134186
355172 -0.051453  -0.010742   -0.118439
355173 -0.043427  -0.018860   -0.110870

[355174 rows x 3 columns]
(355174, 3)

I am open to using another data input method, as long as its not matlab .mat files as I don't have access to it. Possibly it's something to do with how Panda parses the data? If you have time could you upload a simple data input example using something like a txt file with the data files also uploaded please? That would help solve the problem. It may be something clashing on my python install that I can't figure out.