Mayitzin/ahrs #11: float object annot be interpreted as integer (gyro data)

gugiek, 2020-11-28:
Hi, I am trying to get the Madgwick filter working with the adafruit npx IMU and a raspberry pi. 
When I try to run the script I get this error:
  _File "AHRS.py", line 24, in <module>
    gyr = np.ndarray([gyro_x,gyro_y,gyro_z])
TypeError: 'float' object cannot be interpreted as an integer_

Is there something I am not getting? 
I tried transforming the data coming from the sensors into integers, but then I get a new error that says: ValueError: negative dimensions are not allowed
Thanks, code below.

import time
import board
import busio
import adafruit_fxos8700
import adafruit_fxas21002c
from ahrs.filters import Madgwick
import numpy as np

#instantiating sensor and BUS
i2c = busio.I2C(board.SCL, board.SDA)
fxos = adafruit_fxos8700.FXOS8700(i2c)
fxas = adafruit_fxas21002c.FXAS21002C(i2c)

filter = Madgwick()

while True:
    q = np.ndarray([0,0,0,0])
    # Read acceleration & magnetometer.
    accel_x, accel_y, accel_z = fxos.accelerometer
    mag_x, mag_y, mag_z = fxos.magnetometer
    gyro_x, gyro_y, gyro_z = fxas.gyroscope
  
    gyr = np.ndarray([gyro_x,gyro_y,gyro_z])
    mag = np.ndarray([mag_x,mag_y,mag_z])
    acc = np.ndarray([accel_x,accel_y,accel_z])
    
    q = filter.updateMARG(q, gyr, acc, mag)
    print(q)

    time.sleep(0.1)

moutayam, 2020-11-28:
I think if you replace:
```

gyr = np.ndarray([gyro_x,gyro_y,gyro_z])
mag = np.ndarray([mag_x,mag_y,mag_z])
acc = np.ndarray([accel_x,accel_y,accel_z])

```
With
```

gyr = np.array([gyro_x,gyro_y,gyro_z])
mag = np.array([mag_x,mag_y,mag_z])
acc = np.array([accel_x,accel_y,accel_z])

```
It will works, check the difference between np.array and np.ndarray for more information.

gugiek, 2020-11-28:
Thanks for the fast response! 

I tried as suggested but a new error appeared.

 File "AHRS.py", line 27, in <module>
    q = filter.updateMARG(q, gyr, acc, mag)
  File "/home/pi/.local/lib/python3.7/site-packages/ahrs/filters/madgwick.py", l                                                              ine 160, in updateMARG
    qw, qx, qy, qz = q[0], q[1], q[2], q[3]

I think the error might be in my assignment of q, but I tried changing q = np.ndarray([0,0,0,0]) to np.array([0,0,0,0]) and itstill generated an error.

Do you gave any idea what this could be referring to?

moutayam, 2020-11-28:
Quaternion should be calculated from initial roll pitch and yaw, usually you can calculate roll and pitch from accelerometers in static position, and you get yaw from the magnetometer.
Nevertheless, you may use an initial attitude of roll pitch and yaw equal to zero, the corresponding quaternion will be
```
q = np.array([1, 0, 0, 0])
```
Remember that this q is for testing and in real application please calculate the real initial attitude.
You are welcome)

gugiek, 2020-11-28:
Thanks again,

I had also tried:

q = np.array([1, 0, 0, 0]) for testing purposes, but it keeps giving me this error:

File "/home/pi/.local/lib/python3.7/site-packages/ahrs/filters/madgwick.py", line 159, in updateMARG
    q /= np.linalg.norm(q)
TypeError: ufunc 'true_divide' output (typecode 'd') could not be coerced to provided output parameter (typecode 'l') according to the casting rule ''same_kind''

gugiek, 2020-11-28:
so...

Apparently with q = [1, 0, 0, 0] the program does not crash.

Thanks, I'll verify that the numbers make any sense now

moutayam, 2020-11-29:
> Quaternion should be calculated from initial roll pitch and yaw, usually you can calculate roll and pitch from accelerometers in static position, and you get yaw from the magnetometer.
> Nevertheless, [...]

Sorry for my last reply, I give an inaccurate suggestion)
Usually, I use an initial quaternion in a form of a list like the following:
`
q_init = [0.989919, -0.000589, -0.000649, 0.141632]
`
or
`
q_init = [1, 0, 0, 0]
`
Good to hear that you have solved the problem, you may close the issue now.

moutayam, 2020-11-29:
Those websites may help you see a "visualization" of quaternions:
[https://quaternions.online/](url)

Mayitzin, 2020-11-29:
Hi, I'm glad the issues were easily tackled.

For future reference: iterative filters that use angular velocity integration as their main way to estimate attitude (like Madgwick, Mahony, Complementary, etc.) require a valid initial quaternion. Such quaternion must have a norm equal to 1.

The first quaternion used, q=[0, 0, 0, 0], could not be used as initial quaternion, because it didn't actually represent any orientation. Therefore, all other orientations "growing" from it were also invalid.

As @moutayam mentioned, it is preferrable to estimate that initial quaternion using the accelerometer and, if available, the magnetometer. Then, use that estimated orientation as initial orientation.

I will try to add some warnings at the setting of the initial quaternion, so that these issues are spotted sooner.

closed 2020-11-29 by Mayitzin