Mayitzin/ahrs #21: Class constructor with data not working

perotom, 2021-01-04:
I tried to get this simple code working but it seems like the constructor is not working correctly:

```
print(data[GYR].shape)
print(data[ACC].shape)
print(data[MAG].shape)
madgwick = Madgwick(gyr=data[GYR],
                        acc=data[ACC], 
                        mag=data[MAG], 
                        frequency=200)
print(madgwick.Q)
```

will output:

```
(11995, 3)
(11995, 3)
(11995, 3)
AttributeError: 'Madgwick' object has no attribute 'Q'
```

Assumption: In the constructor `self.input = args[0] if args else None` will always be `None` as args is always empty with named parameters?

Mayitzin, 2021-01-04:
Which version are you using?

perotom, 2021-01-05:
Yesterday I did the installation which resulted in: Version: 0.2.2 with Python 3.8.2

orekides, 2021-01-05:
the pypi version is too old, you should install the latest version by following the **readme**

perotom, 2021-01-05:
Ok thanks.
To avoid this it would be better suited for people using pip: `pip install git+https://github.com/Mayitzin/ahrs.git`.

closed 2021-01-05 by perotom