Mayitzin/ahrs #22: pip install from source fails

JackCaster, 2021-01-14:
I have troubles installing the package from source via pip in a new, fresh environment:
```
pip install git+https://github.com/Mayitzin/ahrs.git
```
as I get this error:
```
    ERROR: Command errored out with exit status 1:
     command: 'C:\dev\miniconda\envs\test\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xxx\\AppData\\Local\\Temp\\pip-req-build-rmy_ows9\\setup.py'"'"'; __file__='"'"'C:\\Users\\xxx\\AppData\\Local\\Temp\\pip-req-build-rmy_ows9\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\xxx\AppData\Local\Temp\pip-pip-egg-info-15_9fo6t'
         cwd: C:\Users\xxx\AppData\Local\Temp\pip-req-build-rmy_ows9\
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\xxx\AppData\Local\Temp\pip-req-build-rmy_ows9\setup.py", line 22, in <module>
        from ahrs.versioning import get_version
      File "C:\Users\xxx\AppData\Local\Temp\pip-req-build-rmy_ows9\ahrs\__init__.py", line 11, in <module>
        from . import common
      File "C:\Users\xxx\AppData\Local\Temp\pip-req-build-rmy_ows9\ahrs\common\__init__.py", line 7, in <module>
        from . import mathfuncs
      File "C:\Users\xxx\AppData\Local\Temp\pip-req-build-rmy_ows9\ahrs\common\mathfuncs.py", line 7, in <module>
        import numpy as np
    ModuleNotFoundError: No module named 'numpy'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.                                                        
```
and I suspect the reason is that `versioning.py` is inside the package and it triggers imports of packages not yet installed. (the `setup.py` is importing the package itself).

[Here](https://packaging.python.org/guides/single-sourcing-package-version/#single-sourcing-the-version) there are some suggestions on how to deal with versioning. I believe now the package implements what is described at point 6, which is not recommended. There are some alternative suggestions also [here](https://stackoverflow.com/a/24517154)

Mayitzin, 2021-01-14:
Oh, I hadn't tested the `pip` installation directly from the repository. Thanks for bringing it up. I'll get to it!

Mayitzin, 2021-02-02:
Fixed with commit decf15ee53565ee0aeee39ba0357c20b15101d34.

The versioning is now defined in the `setup.py` by importing from a script in folder `tools` at the top directory. This prevents any try to import third-party modules in a fresh install.

closed 2021-02-02 by Mayitzin