Mayitzin/ahrs #24: Latest version on pypi

sivark, 2021-01-20:
Hello; thanks for a very useful library!

The AngularRate class which I’m trying to use breaks in v0.2 published on pypi, but the refactored version on the master branch seems to work fine. Is there currently a way to download the recent version from pypi? If not, would you consider tagging/publishing the latest version (v0.3) to pypi? Thanks.

Mayitzin, 2021-01-21:
Yes, the newest version is still a release candidate, and I am working on finishing some details (mainly documentation), but as soon as it's ready (hopefully by the end of the month) I will upload it to pypi.

sivark, 2021-02-01:
Thank you very much! I shall look forward to this.

Mayitzin, 2021-02-03:
3df76a9f024e50311e7e2aec35a83fe029303f06 :
Version 0.3 is already in pypi: https://pypi.org/project/AHRS/

closed 2021-02-03 by Mayitzin