Mayitzin/ahrs #31: Quaternion transformation to Roll-Pitch-Yaw angles

Mayitzin, 2021-02-16:
When you see the gyroscope integration, you can see that the Y-axis spins 1 negative turn then 1 positive turn. When you see the estimated and reference orientation, the Yaw simply doesn't follow that. I agree reference Yaw and reconstructed Yaw are similar because their quaternions are very very close. the problem isn't about +-2pi. The problem comes that the `to_angles()`conversion doesn't get you the real movement.

_Originally posted by @anisghaoui in https://github.com/Mayitzin/ahrs/discussions/20#discussioncomment-371173_

aayushsingla, 2021-02-22:
@Mayitzin is this issue confirmed? I am facing a kinda similar issue with Kalman and Mahony. If this issue is confirmed, is there a workaround for this?

anisghaoui, 2021-02-22:
This issue is confirmed to me.  I used Mahony, I got the same observation. My only workaround is to used my phone flat on my hand face up to the ceiling, usb towards my chest. And consider the Z axis as the Yaw. The Yaw is then correct.

The problem is that I checked the current implementation of to_angles() and it seems correct but doesn't have singularities management (which in my case doesn't change a thing).

You can go on an adventure and try to find the quaternion conversion. Gl Hf

anisghaoui, 2021-05-04:
do we have any update on this weirdness?

aayushsingla, 2021-05-04:
No, I switched to a different approach due to the approaching deadlines. This is still unsolved.

> do we have any update on this weirdness?

tt7533, 2021-06-06:
Taking code sample from the documentation

```
attitude = ahrs.filters.Madgwick(acc=acc_data, gyr=gyro_data)
attitude.Q
```

How to convert from quaternion into yaw/pitch/roll angles?

PeterBorisenko, 2023-06-02:
> How to convert from quaternion into yaw/pitch/roll angles?

1. Convert from quaternion to DCM (Rotation Matrix) 
2. Considering a rotation order of Euler Angles, convert DCM to EA.