Mayitzin/ahrs #34: Opposite results from Madgwick and AQUA

Yushchuk6, 2021-02-28:
When used on the same reading Madgwick and AQUA produce opposite results.
This causes spinning in opposite directions.

I can guess that it is just how filters implementation works, but it would be nice if all filters produced the same result.
It should be easy to fix just by flipping the XYZ quaternion result for one on the filters.

Also, didn't test but it is possible that there are more filters that produce one of two results.

Example of rotation:
AQUA
[0.92532097 0.3791205  0.00524886 0.00460348]
Madgwick
[ 0.92497363 -0.3799769  -0.00416885 -0.00489403]

Image example:
[image]

Mayitzin, 2021-03-05:
Thanks for pointing this out. I think I'll make some big benchmark for all estimators and try to find more discrepances like this. Once I find the reason for such an inverted rotation I'll get back to you. Thanks again!

Mayitzin, 2025-02-23:
As mentioned in https://github.com/Mayitzin/ahrs/issues/103 the AQUA Algorithm computes the **inverse quaternion orientation**. This means, with it we compute the orientation of the global reference frame with respect to the local reference frame.

We would need to inverse the orientations, if we want to compare them against the other algorithms.