Mayitzin/ahrs #38: pip install error

indexGui, 2021-05-01:
Hi Mario, 

I'm having troubles installing your package, here attached the error message shortly after the pip install ahrs command:
[image]

i'm dealing with an import error, apparently my raspberry isn't able to find a module called tools.versioning.

how can i deal with it?

thanks, 
Andrea

Mayitzin, 2021-05-07:
Hi,

that seems like a local import error. I guess for the raspberry pi the format has to be changed. I will try to test it and change it accordingly.

indexGui, 2021-05-08:
Thanks,

if you need anything more, just let me know.

cshaw-velo-ai, 2022-12-14:
Isn't the problem here that the tar.gz bundle doesn't include the `tools` directory?  The `ahrs` directory is there, but `tools` is not.

https://files.pythonhosted.org/packages/66/56/7fc17b6c330eff8986e01dfd3744154ffa45cf4ccd45fd779dc6e2713ec1/AHRS-0.3.1.tar.gz