Mayitzin/ahrs #43: How ist measurement noise defined?

AlexVaith, 2021-07-13:
I try to use the extended Kalman filter. I have the datasheet of the sensors being used, but I struggle to identify the specific values the ekf() function is looking for. 

I am a newbie to sensors and datasheets in general. 
Could you help me to identify the values for the following sensor: ICM-20649. 

Thanks in advance

PeterBorisenko, 2023-06-02:
Hello.
Taking a gyroscope datasheet variance is not enough to build a robust estimation. 
This is mostly because datasheet data represents sensor itself and not a system it works onto.
The system (vehicle, tool, etc.) can introduce vibrations which impact sensor's data.

The way I use is the following:
1. sample enough sensor data while all main components of your system are turned on (engines ON and so on) but the system itself is not moving
2. calculate covariance matrix (using a python script or excel)
3. use this covariance matrix in EKF

An example of python code:
```python
import numpy as np
import matplotlib.pyplot as plt
import csv

ACCELEROMETER_DATA_POS= 0
GYROSCOPE_DATA_POS= 3
MAGNETOMETER_DATA_POS= 6

def plotDistribution(dataset1, dataset2, datasetName):
    (ds1Num, ds1Vals)= dataset1
    (ds2Num, ds2Vals)= dataset2
    
    fig, (ax1, ax2)= plt.subplots(1, 2)
    fig.suptitle("{} distribution".format(datasetName))
    
    for i in range(3):
        counts, bins = np.histogram(ds1Vals[:,i] - np.mean(ds1Vals[:,i]))
        ax1.stairs(counts/ds1Num, bins)

    for i in range(3):
        counts, bins = np.histogram(ds2Vals[:,i] - np.mean(ds2Vals[:,i]))
        ax2.stairs(counts/ds2Num, bins)
        
    ax1.grid()
    ax2.grid()
    ax1.set_xlim([-0.15, 0.15])
    ax2.set_xlim([-0.15, 0.15])
    ax1.set_ylim([0, 1])
    ax2.set_ylim([0, 1])
    plt.show()

def printCovarianceMatrix(dataset, name):
    print("{} Covariance matrix:\n{}".format(name, np.cov(dataset.T)))

def getData(filename):
    datafile= open(filename, mode='r')
    reader= csv.reader(datafile)
    rows = list(reader)
    datafile.close()
    
    a= np.empty(3)
    g= np.empty(3)
    m= np.empty(3)

    for row in rows:
        a= np.vstack((a, np.array([float(i) for i in row[ACCELEROMETER_DATA_POS : ACCELEROMETER_DATA_POS+3]])))
        g= np.vstack((g, np.array([float(i) for i in row[GYROSCOPE_DATA_POS : GYROSCOPE_DATA_POS + 3]])))
        m= np.vstack((m, np.array([float(i) for i in row[MAGNETOMETER_DATA_POS : MAGNETOMETER_DATA_POS + 3]])))

    a= a[1:]
    g= g[1:]
    m= m[1:]
    num_samples= len(rows)
    timeline= np.linspace(0,num_samples,num=num_samples)
    return {'n':num_samples, 'a':a, 'g':g, 'm':m, 't':timeline}

sensData= getData('./SensorNoise.csv')

printStandardDeviation(sensData['a'], "Accelerometer")
plotDistribution((sensData['n'], sensData['a']),
                 (sensData['n'], sensData['a']), 'Accelerometer noise')
printStandardDeviation(sensData['g'], "Gyroscope")
plotDistribution((sensData['n'], sensData['g']), 
                 (sensData['n'], sensData['g']), 'Gyroscope noise')
printStandardDeviation(sensData['m'], "Magnerometer")
plotDistribution((sensData['n'], sensData['m']), 
                 (sensData['n'], sensData['m']), 'Magnerometer noise')
```

Hope it'll be useful

Mayitzin, 2025-02-23:
There are two steps in the KF: the **prediction step**, and the **correction step**. Two types of noise are defined for each of the steps:

- Process Noise, represented by $\mathbf{\sigma}_\omega$, is involved in the prediction step.
- Measurement Noise, represented by $\mathbf{\sigma}_z$, is involved in the correction step.

where $\mathbf{\sigma}_\omega$ is built from the gyroscope data,

```math
\mathbf{\sigma}_\omega = \mathbf{\sigma}_\mathrm{gyro}
```

can be built from either the accelerometer noise only:

```math
\mathbf{\sigma}_z = \mathbf{\sigma}_\mathrm{acc}
```

or from the accelerometer and magnetometer:

```math
\mathbf{\sigma}_z = [\mathbf{\sigma}_\mathrm{acc}, \mathbf{\sigma}_\mathrm{mag}]
```

These are the parameters expected in the class `EKF`.

They are responsible for describing the amount of expected noise in a system, and are used to build the covariance matrices `Q` and `R`.

The technique of estimating these values is called **Tuning**, and once the model is defined, it is the most important step to perform.

The most common approach is to have a known ground truth and constantly adjust them until the RMS is minimized.

For more details about the process check:

https://www.mathworks.com/help/fusion/ug/tuning-kalman-filter-to-improve-state-estimation.html

closed 2025-02-23 by Mayitzin