Mayitzin/ahrs #48: symbo

gooyle, 2021-08-06:


gooyle, 2021-08-06:
thx for your package！！！it helps a lot,i think here are some mistakes, i use xsens data to test.
in file:
> ahrs/ahrs/filters/mahony.py

>#Rotate magnetic field to inertial frame
~~h = R@m~~     
 h =m@R
~~v_m = R.T@np.array([np.linalg.norm(-[h[0], h[1]]), 0.0, h[2]])~~   
 v_m = R.T@np.array([np.linalg.norm([h[0], h[1]]), 0.0, h[2]])
and here is the results:
[image]

Mayitzin, 2022-01-28:
This was adressed already in the commit 34f959f96adba5064d1e4f0142eaad971f7f3c64, and there was a major change with it too.

The creation of the `v_m` was, apparently, not correct. Please, test the new version, if possible, and let me know if you see any improvements at all.

closed 2024-09-16 by Mayitzin