Mayitzin/ahrs #50: Error in EKF Measurement Model Docs

maximilian-nitsch, 2021-09-09:
Hi,

I think I maybe found multiple (sign) errors in the EKF measurent model:
https://ahrs.readthedocs.io/en/latest/filters/ekf.html

In for example in the first row of h(qhat_t), the sign before gx*(1/2-qy^2-qz^2) has to negative: -gx*(1/2-qy^2-qz^2).
I have calculated this both manually and with the Symbolic Toolbox of MATLAB and come to the same result. 

You can check my calculation by defining your gravity vector g = [gx gy gz]^T and mutliply from left with rotation matrix C(qhat).

Is this a calculation error or have I overlooked something?

Best regards
Max

Mayitzin, 2021-09-13:
Hi Max,

Thanks for reviewing the documentation. The line in the documentation is, on my opinion, still correct, as it is the product of the _transposed_ rotation matrix **C**. That is why the following element has also opposed signs (`gz*(qx*qz−qw*qy)`)

In the package it can also be confirmed at line [1198](https://github.com/Mayitzin/ahrs/blob/master/ahrs/filters/ekf.py#L1198) of the script `ekf.py`. There we are using the transposed rotation matrix, as it states in the documentation.

In the end, one could not even see the effects of this product, because after multiplying the transposed rotation matrix **C** with the gravitational acceleration vector **g**=[gx, gy, gz]=[0, 0, 1], it yields only elements multiplying _gz_, Even changing between NED or ENU configurations (they just flip the sign of the resulting vector)

Do you think my conclusion is correct? If not, please let me know so that I can fix any further problem. Thanks!

Mayitzin, 2021-09-13:
Maybe I should mention in the documentation that we are transposing the rotation matrix because we are comparing the vectors in the sensor frame (as opposite to the global frame) and, thus, an opposite rotation is needed.