Mayitzin/ahrs #55: Yaw static error

AliaChe, 2021-12-13:
Hello, 

Thank you for this very useful code. 

I'm facing an issue with the yaw estimation using Madgwick/Mahony filters. 

I povide ideal sensors measures in ENU frame : 
- aX = [0.0] * n
- aY = [0.0] * n
- aZ = [9.81] * n                  # m/s^2
- gX = gY = gZ =[0.0] * n    # sensor is static
- mX = [0.0] *n 
- mY = [0.05] * n                 # mT
- mZ = [-0.05] *n

Where n is the number of samples set to:
- n = 10000
The sampling frequency is set to:
- Fs = 100

With these inputs, I get  null roll, pitch and yaw. I assume then both filters are providing the orientation in ENU frame.

The issue starts when I add some noise to the gyroscope measures, a noise of mean zero and standard deviation of 0.01:
gX += np.random.normal(0, 0.001, n) 
gY += np.random.normal(0, 0.001, n)
gZ += np.random.normal(0, 0.001, n)

The estimated yaw presents a static error of -pi/2 with Madgwick and pi/2 with Mahony.
Tuning the filters doesn't  fix the issue. 

Any ideas? 

Thank you

dapowan, 2022-04-07:
The same to me. Madgwick/Mahony has significant yaw errors when adding noises to gyroscope. And the EKF filter cannot output the same results even when all gyroscope measurements are equal to 0. The quaternion keeps changing over time.

AliaChe, 2022-04-21:
Hello, 

Found out what I was doing wrong: 
- set the gyroscope measures to (zeros + epsilon) instead of zeros.
- Analyze the results as follows : Madgwick results are in NWU frame, while Mahony returns the attitude in the SEU

Mayitzin, 2022-04-21:
I think your last bulletpoint is something I have to take a look at. All the algorithms have been developed independently by their authors and might have different reference frames, as you point out. I will have to think of a way to deal with it. Thanks for bringing it up!