Mayitzin/ahrs #73: Error in Madgwick updateMARG "unexpected keyword argument" when using dt

viatto, 2022-05-03:
when I try to use this function the command window alerts me of the argument doesn't exist even if it is mentioned in the documentation. 
q =  madgwick.updateMARG( q=qprec, gyr=g,acc=a, mag=MagVals, dt=deltat)
In addiction when I tried to use without dt argument I have the following error:
 q =  madgwick.updateMARG( q=qprec, gyr=g,acc=a, mag=MagVals)
  File "/Users/vittoriaattolini/opt/anaconda3/lib/python3.8/site-packages/ahrs/filters/madgwick.py", line 668, in updateMARG
    h = q_prod(q, q_prod([0, *m], q_conj(q)))               # (eq. 45)
  File "/Users/vittoriaattolini/opt/anaconda3/lib/python3.8/site-packages/ahrs/common/orientation.py", line 64, in q_conj
    if q.ndim>2 or q.shape[-1]!=4:
AttributeError: 'list' object has no attribute 'ndim'

I followed the documentation indication: qprec is an array of 4 values, g, a and Magvals are array of 3 values

Mayitzin, 2022-05-09:
Hi viatto,

both errors you mention origin from the same submodule `orientation`, where the function `q_conj` assumed a `numpy.array` was given, which has the attributes `ndim` and `shape´.

When a list is given, it would raise an Error, because lists do not contain these attributes.

I fixed it with the commit 5d9016b4b712be12f77ba0ab3c18a7d8e2be53c2 by creating a copy of the given input array (`q`) and using this copy instead.

Please give it a try again, and let me know if it helped.

Thanks!

viatto, 2022-05-10:
Thank you! Now it's working!

closed 2022-05-10 by viatto