Mayitzin/ahrs #99: Madgwick Reference Frame

bdeming155, 2023-08-07:
@Mayitzin can you please tell me what rotation convention the output of the Madgwick filter represents and what reference frame it uses? 

For example is it the rotation from the NED frame to the body frame? Is it the rotation from the body frame to the ENU frame? Left vs right hand coordinate system? I am having trouble figuring out what reference frame the output is in. Also some more documentation specifying this would be super helpful. Thanks!

Mayitzin, 2023-08-22:
Hi, the reference frame used in Madgwick's is right-handed NED. I updated the information in the documentation with the commit 97609e0256a32b0af6d08ec7f577a9ca62e6e0d7.

I hope this helps to clarify things.

bdeming155, 2023-08-22:
Thank you that clears things up!

closed 2023-10-04 by Mayitzin