Mayitzin/ahrs #106: Weird behavior in to_DCM and from_DCM in Quaternion class

Floabsolut, 2024-01-19:
First of all, Thanks for making this package available!

While using the Quaternion class I experienced weird behavior to go to / from DCM.

For example:
1. if your satellite is at lat=0, lon=0 and velocity is [0,0,1] (Going north). Then the DCM in ECEF is defined as 
dcm_ecef2platform = np.array([x_platform, y_platform, z_platform])
with 
y_platform = velocity = [0,0,1]
z_platform = pointing = [-1, 0, 0]
x_platform = cross(y_platform, z_platform) = [0, -1, 0]
so DCM = array([[ 0., -1.,  0.],       [ 0.,  0.,  1.],       [-1.,  0.,  0.]])
If I use the method from_DCM I have different quaternions according to the method parameter I choose.
The answer could also be obtained with the roll pitch yaw angles:
Quaternion().from_rpy(np.array([-90, 0, 90]) * DEG2RAD) which gives the correct quaternion: Quaternion([0.5, 0.5, -0.5, -0.5]).

2. The method to_DCM gives unexpected output some times.
for Quaternion([ 0.5  0.5 -0.5 -0.5]) the to_DCM gives [[ 0.  0. -1.] [-1.  0.  0.] [ 0.  1.  0.]] instead of [[ 0. -1.  0.] [ 0.  0.  1.] [-1.  0.  0.]]

Maybe I am missing something, I would be more than happy to help there.

Floabsolut, 2024-01-19:
Maybe from_DCM could be :

```bash
    @staticmethod
    def from_DCM(mat):
        """ Returns the quaternion corresponding to the unitary DCM matrix. """
        mat = np.array(mat)
        tr = np.trace(mat)
        d = 1 + 2 * mat.diagonal() - tr
        qsquare = 1 / 4 * np.array([1 + tr, d[0], d[1], d[2]])
        qsquare = qsquare.clip(0, None)  # avoid numerical errors
        # compute signs matrix
        signs = np.sign([mat[1, 2] - mat[2, 1], mat[2, 0] - mat[0, 2], mat[0, 1] - mat[1, 0],
                         mat[0, 1] + mat[1, 0], mat[2, 0] + mat[0, 2], mat[1, 2] + mat[2, 1]])
        signs_m = np.zeros((4, 4))
        signs_m[np.triu_indices(4, 1)] = signs
        signs_m += signs_m.T
        signs_m[np.diag_indices(4)] = 1.
        # choose appropriate signs
        max_idx = qsquare.argmax()
        coords = np.sqrt(qsquare) * signs_m[max_idx]
        return Quaternion(*coords)
```

Floabsolut, 2024-01-19:
I have the impression to_DCM is giving the Transpose of what DCM should be.

Mayitzin, 2024-09-09:
Hi,

sorry for the late reply. It can be possible that the reference frames are considered opposite in the description of the attitude. So, instead of describing a local frame with respect to the global, it might be describing a global frame with respect to the local frame (the opposite rotation.)

The computation of the Direction Cosine Matrix from a given quaternion, $q=(q_w, q_x, q_y, q_z)$, is computed as defined in [Wikipedia](https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation#Quaternion-derived_rotation_matrix):

$$
\\mathbf{R}(\\mathbf{q}) =
\\begin{bmatrix}
1 - 2(q_y^2 + q_z^2) & 2(q_xq_y - q_wq_z) & 2(q_xq_z + q_wq_y) \\\\
2(q_xq_y + q_wq_z) & 1 - 2(q_x^2 + q_z^2) & 2(q_yq_z - q_wq_x) \\\\
2(q_xq_z - q_wq_y) & 2(q_wq_x + q_yq_z) & 1 - 2(q_x^2 + q_y^2)
\\end{bmatrix}
$$

You can also try this example online with Matlab's [rotmat](https://mathworks.com/help/nav/ref/quaternion.rotmat.html) function and it will give you the same result as with AHRS' function:

```
quat = [0.5 0.5 -0.5 -0.5];
rotm = quat2rotm(quat)

  rotm = 
         0     0    -1
        -1     0     0
         0     1     0
```

For the conversion from DCM to Quaternion you can use the built-in parameter `dcm` of the class `Quaternion`:

```python
>>> R = ahrs.DCM(np.array([[ 0., 0., -1.], [-1., 0., 0.], [ 0., 1., 0.]]))
>>> ahrs.Quaternion(dcm=R)
Quaternion([ 0.5,  0.5, -0.5, -0.5])
```

or `DCM`'s method `to_quaternion`:

```python
>>> R.to_quaternion()
array([ 0.5,  0.5, -0.5, -0.5])
```

I see that you are using [Shepperd](https://arc.aiaa.org/doi/10.2514/3.55767b)'s method to compute a quaternion from a DCM. This you can already do by setting `method` equal to `shepperd` in the method `to_quaternion`:

```python
>>> R.to_quaternion(method='shepperd')
array([ 0.5, -0.5,  0.5,  0.5])
```

But, as you can see, the signs are different, representing a different attitude. May I suggest to use [Chiaverini](https://www.researchgate.net/profile/Bruno-Siciliano-2/publication/262391661_The_Unit_Quaternion_A_Useful_Tool_for_Inverse_Kinematics_of_Robot_Manipulators/links/548da3550cf225bf66a5c2f3/The-Unit-Quaternion-A-Useful-Tool-for-Inverse-Kinematics-of-Robot-Manipulators.pdf)'s method? I empirically found it to be more consistent. Thus, it is the default method:

```python
>>> R.to_quaternion()
array([ 0.5,  0.5, -0.5, -0.5])
>>> R.to_quaternion(method='chiaverini')
array([ 0.5,  0.5, -0.5, -0.5])
```

There are 5 different methods for this conversion. See [to_quaternion](https://ahrs.readthedocs.io/en/latest/dcm/classDCM.to_quaternion.html)'s method documentation to check them all.

Finally, regards the Earth-Centered Earth-Fixed frame. Could we know the directions of the local frame? When it is East-Nort-Up, it would describe something different against North-East-Down. Maybe we can start attacking the problem from there.

Thanks for the feedback, and let me know if I can assist you with anything else.

Mario.

Mayitzin, 2024-09-09:
I retract my words. Shepperd's is the BEST method. I refactored it to be consistent with the original description and it is simply the most precise out of all of them.

I also set it already as the default method in classes `DCM`, `Quaternion`, and `QuaternionArray`.

Mayitzin, 2024-09-16:
The transformations from DCM to quaternion have been updated and documented. Extra unit tests were added with commit 595ab9877c77f5960e39b352c0ef551d2d2efcb9

I hope you don't have further problems with it. Otherwise, please write back.

Mario.

closed 2024-09-16 by Mayitzin