Mayitzin/ahrs #107: Madgwick with asynchronous sensors

bdeming155, 2024-02-04:
@Mayitzin I would love to hear your thoughts on the following problem:

I have an accel, gyro and mag and want to use the Madgwick filter to estimate orientation. My accel and gyro are synced and running at the same rate but my mag is a separate sensor running at a lower rate. 

Can I call both updateIMU() and updateMARG() on the same instance of Madgwick()? I basically want to call updateIMU() with the intermediate accel and gyro measurements and then when a new mag measurement comes in call updateMARG(). 

I feel like this would work fine since we pass the quaternion estimate from the last time step and the delta_t between measurements to the update functions but would love to hear your thoughts. Thanks!

Mayitzin, 2024-02-06:
This is a common struggle with many applications. And in my opinion you can go on as you explained it.

The way Madgwick works (and many other algorithms) is by computing the attitude with the gyroscope first (simple integration), and improving this estimation with any other available sensor.

When `updateIMU()` is called it improves the estimation with the accelerometer.

When `updateMARG()` is called it improves the estimation with the accelerometer AND magnetometer.

I see no problem if you want to use `updateMARG()` asynchronously.

bdeming155, 2024-02-06:
That makes sense, thanks for the reply!

closed 2024-02-06 by bdeming155