Mayitzin/ahrs #114: Need pre-recorded data to test the AHRS library's Extended Kalman Filter

amanabt, 2024-06-11:
Friends,

I am just getting started with a project where I have to implement an Extended Kalman Filter (EKF) based on the EKF implementation of this library. This is the first time I am venturing into this field and need some help getting started.

At first I wanted to get a feel of how the your implementation of your EKF is used. Unfortunately, I do not have a primary source (hardware) of getting the gyroscope and accelerometer data.

Could you please suggest me some pre-existing data that I could use to test this library? I need 3 axis gyroscope data and 3D accelerometer data.

Mayitzin, 2024-06-13:
Hi, @amanabt 

In the discussions post of [Datasets](https://github.com/Mayitzin/ahrs/discussions/20) you can find a couple that are good to test the different algorithms, including the EKF.

Cheers!

Mayitzin, 2024-09-14:
No further comments were added. I think we can close this issue.

closed 2024-09-14 by Mayitzin