Mayitzin/ahrs #115: EKF, Madgwick possible for acc, gyr [mag] data from different timestamps?

johann-petrak, 2024-06-12:
As I understand the acc, gyr and optional mag data are assumed to get measured 

* at regular intervals (as expressed by the sampling frequency parameter 
* all at the same time

What to do if neither of these conditions is fulfilled, e.g. the data arrives at non-regular intervals, the data for acc may arrive at a different time than that of the gyro and maybe the frequencies are even different for acc, gyr and mag? 

Is there anything that can be done using this library? Are there alternate solutions for doing this from Python?

Mayitzin, 2024-06-13:
That's a very interesting problem. Unfortunately the asynchronous data will make it difficult to properly estimate an orientation.

However, I'd try to interpolate the data to the same timestamp using [splines](https://en.wikipedia.org/wiki/Spline_interpolation). Check the 1D example of this implementation using Scipy:
https://docs.scipy.org/doc/scipy/tutorial/interpolate/ND_unstructured.html#d-example

If you want to use the "latest" data, and some sensors are lagging behind, I'd try to extrapolate them with Cubic Splines, which, in my opinion, seem more natural reactions than simply linear extrapolations or data padding. Check also the extrapolation tricks in Scipy's examples:
https://docs.scipy.org/doc/scipy/tutorial/interpolate/extrapolation_examples.html

johann-petrak, 2024-06-13:
Thank you that is very helpful! 

My (limited) understand was that the different timestamps my not be in principle a problem with KF and EKF, as it would only influence for which points in time the predictions have to be made so that the update step can act on them, but for the Madgwick algorithm I really do not know.

I think interpolating the data could be a good idea if the data itself does not change too much between the sample points (which is sadly another problem with the data we have) but it is definitely worth a try!