Mayitzin/ahrs #117: Is the `meridian_curvature_radius` formula correct?

andrei-ng, 2024-07-11:
I am not sure if it is a bug or I am missing something myself. 

Botht the code and the documentation for the [`meridian_curvature_radius`](https://ahrs.readthedocs.io/en/latest/wgs84.html#ahrs.utils.wgs84.WGS.meridian_curvature_radius) use the cubic root for the denominator. 

All the references I checked use the formula with the power `3/2` for the denominator. E.g., https://en.wikipedia.org/wiki/Earth_radius#Meridional or  [AIDED Navigation - GPS With High Rate Sensors, Jay Farrell](https://books.google.nl/books?id=yNujEvIMszYC&lpg=PP1&pg=PR3#v=onepage&q&f=false) formula 2.6 page 32. 

If I misuderstood the the meaning of the formula, feel free to close this issue.

Mayitzin, 2024-07-11:
Hi, that seems very strange indeed. I took the formula from the official WGS84 Report of the US Office of Geomatics.

You can find it here: https://earth-info.nga.mil/index.php?dir=wgs84&action=wgs84

The formula is found in page 68 (labeled page 7-8), where it uses the cubic root without squaring the denominator.

However, based on the references you provide I am inclined to agree with your comment. For example, the Wikipedia page takes the formula from a document written by Christopher Jekeli from the Division of Geodetic Science at the Ohio State University.

You can find that document here: https://kb.osu.edu/server/api/core/bitstreams/404dbfb8-da94-5f09-baf4-dbee438ece83/content

In that document the formula of the curvature of the meridian ellipse (equation 2.37) is defined, as you mentioned, at page 32. I followed his definition, and it seems to arrive to the correct formula.

To further solidify your comment, I found more definitions that agree with that one:

- [Grid, ground, and glove: Distances in the GPS era](https://www.researchgate.net/publication/289101489_Grid_ground_and_globe_Distances_in_the_GPS_era) by Thomas H Meyer (page 4.)
- [Radius of Earth - Radii used in Geodesy](https://www.oc.nps.edu/oc2902w/geodesy/radiigeo.pdf) by James R. Clynch (page 4.)
- [Global Positioning Systems, Inertial Navigation, and Integration](https://www.globalspec.com/reference/14801/160210/chapter-9-4-3-inertial-systems-technologies-earth-models) by Mohinder S. Grewal et al (equation 9.26)

So, I dare to say you have found an error in the official US Government documentation from 2014.

If you agree with my observations, I would change the formula, and suggest you to notify the US Office of Geomatics that they have made a mistake.

andrei-ng, 2024-07-12:
Hi Mario,

Thanks for checking and for all the references you provided! Indeed, I was quite puzzeled as well since all the references I found had the 3/2 power. 

I will reach out to the US Office of Geomatics via email and inform them that we think the document is incorrect.

Mayitzin, 2024-07-16:
The computation of [meridian_curvature_radius](https://ahrs.readthedocs.io/en/latest/wgs84.html#ahrs.utils.wgs84.WGS.meridian_curvature_radius) (and its documentation) has been updated with commit d117e876bf969158a6bab169dfb1d80aeeb1c946

closed 2024-07-16 by Mayitzin

andrei-ng, 2024-07-17:
@Mayitzin, FYI, I got confirmation from US Office of Geomatics that indeed the document has an error and that they also recently discovered it. They will be publishing a Version 2 of the document with more updates sometime in Q4 of of this year.