Mayitzin/ahrs #118: Please add ahrs.__version__

johann-petrak, 2024-07-16:
It is a very useful Python convention to be able to ` import ahrs`  then use `ahrs.__version__`  to find the version of the installed package.

Mayitzin, 2024-07-16:
Hi! You can still use it. It is still defined in the root `__init__`.

I've tested it in a new virtual environment and you can still call it.

```python
>>> import ahrs
>>> ahrs.__version__
'0.4.0'
```

johann-petrak, 2024-07-21:
Does not work for me after doing `pip install -U ahrs`  so maybe this is not in a released version yet?

```
(erdprofi) johann@johann:~$ pip install -U ahrs
Requirement already satisfied: ahrs in ./miniconda3/envs/erdprofi/lib/python3.11/site-packages (0.3.1)
Requirement already satisfied: numpy in ./miniconda3/envs/erdprofi/lib/python3.11/site-packages (from ahrs) (1.23.5)
(erdprofi) johann@johann:~$ python
Python 3.11.9 (main, Apr 19 2024, 16:48:06) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ahrs
>>> ahts.__version__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ahts' is not defined. Did you mean: 'ahrs'?
>>> 
```

Mayitzin, 2024-07-21:
Ah, two points here:

1. The newest version in the repository is not yet updated in pip. So, it should work as it has in the last months.
2. According to your error traceback, you're trying to import `ahts`, and not `ahrs`.

Let me know if you're still having trouble.

johann-petrak, 2024-07-21:
Thanks, I was assuming after your comment that it was just a matter of the latest version not being on pypi :)

And sorry, yes, that was a typo, but it does not change with the correct import of course. 

I tend to stick to release versions or packages for the stuff I develop ...

closed 2024-07-21 by johann-petrak