Mayitzin/ahrs #122: Wrong implementation of DCM.log()

lucagrementieri, 2024-09-11:
The new implementation of `DCM.log()` introduced with [this recent commit](https://github.com/Mayitzin/ahrs/commit/0811bc3b15c123c4e1574a322ead6e8527ec7428) outputs coefficient that don't match the result of `scipy.linalg.logm()` for rotations with an angle higher than 90°.

```
>>> import ahrs
>>> import scipy.linalg
>>> import numpy as np
>>> 
>>> r = ahrs.DCM(axang=([0, 0, 1.0], 0.75 * np.pi))
>>> print(r.log)
[ 0.         -0.          0.78539816]
>>> print(scipy.linalg.logm(r).round(6))
[[-0.       -2.356194  0.      ]
 [ 2.356194 -0.        0.      ]
 [ 0.        0.        0.      ]]
```

The problem comes from [this line](https://github.com/Mayitzin/ahrs/blob/2a474ac5244479c2758c36e5796af755a7030566/ahrs/common/dcm.py#L656): there the arcsin of a norm, i.e. a positive number, is always a number in [0, π/2] radians so it cannot recover angles in [π/2,π]. The sign of the cosine is necessary to recover the correct angle and disambiguate between the angles α and π-α that have the same sine.

Mayitzin, 2024-09-11:
Hi, Luca.

You're right. The new commit was a wrong definition. I reverted it to the correct case and added a guard for when the matrix is the trivial case of the identity. Besides, I added a reference, where a deeper explanation of the log(R) can be found.

The changes are up with the new commit dff1b807da37058ad8c93302f605d43efa888494

If you experience something off, let me know.

Mario.

lucagrementieri, 2024-09-11:
Thank you for the correction!

closed 2024-09-11 by lucagrementieri