Mayitzin/ahrs #125: adaptive gain seems not correct in aqua algorithms

whoispo, 2024-10-15:
[image]

if alpha equals 0 then, alpha will be 0 forever.   Because in the code, alpha updates:
[image]

Even if f equals 1, then new alpha will be 0.

I have read the reference article, the gain is not stored and used again.  
[image]

Mayitzin, 2024-10-15:
You are right. The gain wasn't being properly updated.

The commit de6e3632055a57c45b1d66b21113174e66dad37e has modified this part and its calls within the algorithm. The changes are:

- The final computation is made with an optional alpha value named `alpha_bar` (equal to 0.1) to be consistent with the original article.
- Because it is optional it avoids the problem of being called with the predefined `self.alpha` as it was.
- As mentioned, the alpha gain is re-defined in the function `adaptive_gain()` as the second parameter now. The first parameter is now the **Measured local acceleration vector**, as it always should have been.
- The calls to `adaptive_gain()` in the methods `updateIMU()` and `updateMARG()` are done now with the measured acceleration vector only.
- The computation of the **gain factor** $f(e_m)$ has been simplified using NumPy's function [clip](https://numpy.org/doc/stable/reference/generated/numpy.clip.html).
- The documentation is updated accordingly using $\overline{\alpha}$ for the constant value that gives the best filtering result in static conditions.

If you have further observations, let me know.

Mayitzin, 2024-10-30:
Unit tests for the function `adaptive_gain` were added with commit fd4ad3b56a0115fabda2b7f5d17453ab5c4c94b5

I think we can close this issue.

closed 2024-10-30 by Mayitzin