Mayitzin/ahrs #131: Is it possible to use varied data difference instead of fixed Dt or frequency in updateIMU()?

AgentEgeus, 2025-02-19:
My IMU data has ticks from an MCU which measures time difference in milliseconds. Due to processing time, this time difference is not constant. I tried to use filters in ahrs such as the Mahony and Madgwick filters, but both of them require fixed frequencies in the constructor, or Dt in the function of updateIMU().

Could you point me to the documentation whether or not varied time differences is supported in updateIMU()?

Thank You

Mayitzin, 2025-02-20:
Hi! Yes, it is possible to set the value of Dt at every iteration **as long as you have a time-stamp of each measurement**.

For that you have to call the method `updateIMU()` each time with a different Dt. By default it uses the fixed time difference set at the construction, but this is overwritten if set in the call.

Let's assume you have an array of sensor data, where each row is a sample with a time-stamp, like:

```python
[[0.0100, 0.1236, 0.2346, 0.1342, 0.9191, 0.0108, 0.0346],
 [0.0200, 0.1232, 0.2349, 0.1344, 0.9129, 0.0122, 0.0371],
 [0.0350, 0.1237, 0.2342, 0.1347, 0.9176, 0.0147, 0.0323],
 [0.0400, 0.1233, 0.2345, 0.1340, 0.9133, 0.0170, 0.0369]]
```

The first value of each line represents the time-stamp, the following three are gyroscope values, and the last three are accelerometer values.

If we compute the time difference, a.k.a. time step, between the samples we can quickly see that the values are not sampled at a constant rate.

> dt_1 = 0.020 - 0.010 = 0.01
> dt_2 = 0.035 - 0.020 = 0.015
> dt_3 = 0.040 - 0.035 = 0.005

However, the method `updateIMU()` still deals with it.

An example with the Mahony filter would be:

```python
quaternions = [[1, 0, 0, 0]]
mahony = ahrs.filters.Mahony(q0=quaternions[0])
time_stamp = 0.0
for row in data_array:
    new_time = row[0]
    time_difference = new_time - time_stamp
    new_quaternion = mahony.updateIMU(q=quaternions[-1], gyr=row[1:4], acc=row[4:7], dt=time_difference)))
    quaternions.append(new_quaternion)    # append new quaternion at the end of our list of quaternions
    time_stamp = new_time
```

This will estimate the new orientation with a different time step every time. Let's go line by line.

- First, we build a 2d-array (a list of lists) with only one value in it (a single list inside another): the initial quaternion. For our case we assume it is the origin, or the unit quaternion `(1, 0, 0, 0)`.

```python
quaternions = [[1, 0, 0, 0]]
```

- After that we build our Mahony object and we initialize it with the known initial quaternion. The only value in our 2d-array:

```python
mahony = ahrs.filters.Mahony(q0=quaternions[0])
```

- We don't know if the samples are retrieved at a constant frequency, but we have a time stamp for every new set of sensor measurements. We assume it starts from zero, so we set a variable `time_stamp` equal to zero:

```python
time_stamp = 0.0
```

- Now it comes the fun part. For every new sample after it, we compute the difference between the current `time_stamp` and the previous one:

```python
for row in data_array:
    new_time = row[0]
    time_difference = new_time - time_stamp
```

- This time difference is the value `dt`, and you have to use it when calling the method `updateIMU()`. **CAREFUL**: This value is, by default, in seconds. If you are using milliseconds, as might be your case, you would have to multiply it by `1000.0`

```python
new_quaternion = mahony.updateIMU(q=quaternions[-1], gyr=row[1:4], acc=row[4:7], dt=time_difference)
quaternions.append(new_quaternion)    # append new quaternion at the end of our list of quaternions
```

- Finally, you save the newest time stamp as the current time stamp.

```python
time_stamp = new_time
```

And you continue in the loop until there are no more samples to process.

This way you estimate all attitudes with the correct time difference.

The same applies for the method `updateMARG()`, and for the methods of class `Madgwick`.

I hope I could help you to understand the implementation of the estimators. If there are further questions, please let me know.

Mario.

AgentEgeus, 2025-02-21:
Thank you so much for the quick turnaround! I will try this during the weekend and let you know how it goes.

AgentEgeus, 2025-02-22:
After upgrading to 0.4.0 version, I can use the dt argument now. 

Thank you again for all your help and the detailed example and quick response.

Mayitzin, 2025-02-23:
Glad I could help. I'll finish up some details to finally release 0.4 in the PyPI.

closed 2025-02-23 by Mayitzin