Mayitzin/ahrs #132: Question on the EKF formula implementaion

Ynjxsjmh, 2025-02-28:
I have a question on the implementation of [h](https://ahrs.readthedocs.io/en/latest/filters/ekf.html#ahrs.filters.ekf.EKF.h) and [H](https://ahrs.readthedocs.io/en/latest/filters/ekf.html#ahrs.filters.ekf.EKF.dhdq) in Measurement Model.

The derived formula in the doc is related with `q` and `gyr`. However, the implementation only uses `q`.

https://github.com/Mayitzin/ahrs/blob/master/ahrs/filters/ekf.py#L1248-L1251

https://github.com/Mayitzin/ahrs/blob/master/ahrs/filters/ekf.py#L1321-L1324

Mayitzin, 2025-02-28:
Hi, I think it is a confusion with the terms. The measurement model `h`, and its Jacobian `H`, are defined using the quaternion `q`, and the measured acceleration, represented with `g` (for gravity).

In the code we set the names of the gravitational term as `a_ref`, because it represents the "reference acceleration values."

The same goes for the magnetic reference being `r` in the documentation, and `m_ref` in the code.

I hope this helps to clear the confusion.

Ynjxsjmh, 2025-03-01:
Thanks for your clarification, it solves my confusion.

closed 2025-03-01 by Ynjxsjmh